OLIMEX/ESP32-POE #2: PoE output power specification

mangtronix, 2019-02-06:
What is the amount of power that can be drawn from the PoE module?  I looked and couldn't find any specification.  Something like 5W at 5V?  It looks like the output is fixed at 5V?

I want to attach some NeoPixel-style LED strips. Thanks!

DanKoloff, 2019-02-06:
We have tested up to 800mA at 5V.  This is the safe threshold. 5W at 5V might be possible but depends on other load and the unit would probably heat up.

jcstaudt, 2019-02-18:
FYI: I noticed these relevant power output concerns listed on [Crowd Supply](https://www.crowdsupply.com/silicognition/wesp32).

> Olimex provides no specification on available output power. The inductor specified in the schematic is specified for a maximum of 2 A, which, at 5 V, would limit the available power to a maximum of 10 W. The classification resistor in the schematic would indicate 12.95 W max, but between the power inductor limits and the lack of cooling features on the circuit board, achieving this output power is very unlikely.

TsvetanUsunov, 2019-02-18:
When you build network of many sensors, you try to lower your power consumption as much as possible. What is the reason to make one node which to suck all available power from the POE hub?

mangtronix, 2019-02-18:
I’d like to build a network of lights. Using a 12W RGBW LED would be great. It’s a different application than a sensor network.

TsvetanUsunov, 2019-02-18:
We definitely can't cover all use cases, when we design something we try to balance the features most customer would need and keep budget low. You can choose other solution like the x3 times more expensive Crowd Supply solution quoted above for your appication.

mangtronix, 2019-02-18:
No problem, if the spec sheet said "tested at 4W" I would have known. Best luck with your products!

jcstaudt, 2019-02-18:
@TsvetanUsunov Sorry, I didn't intend to make a business case for one product or another with my previous post. I found some tested information that might help fill in some gaps and potentially help in designing a power output specification, and I thought it would help add to the discussion.

closed 2019-04-09 by DanKoloff