OLIMEX/ESP32-POE #3: Power isolation for IEEE 802.3at compliance

jcstaudt, 2019-02-18:
I stumbled across a criticism of the ESP32-POE and wanted to submit a project issue for awareness and potential resolution:

> The Olimex board has some major cons [...] for those who want a PoE board for commercial installations.

> No isolation. The IEEE 802.3at spec demands isolation up to 1500 Vrms and the product page of the Olimex ESP32-POE specifically mentions that there is no isolation and that you should not have Ethernet and USB plugged in simultaneously.

Source: [Crowd Supply](https://www.crowdsupply.com/silicognition/wesp32)

TsvetanUsunov, 2019-02-18:
Yes, I read this, the guy was desperately to justify his high price and I didn't want to spoil his pitch and campain.
Please show me where 802.3at state 1500Vrms insulation is necessary?
Making insulation adds parts to the board, which in most of the cases people would not need.
ESP32-POE works in field connected mostly to sensors or relays etc. There is no need for such insulation in most of the cases.

jcstaudt, 2019-02-20:
> **33.4  -  Additional Electrical Requirements**
> 33.4.1 - Isolation

> PDs and PSEs shall provide isolation between all accessible external conductors, including frame ground (if any), and all MDI leads including those not used by the PD or PSE. Any equipment that can 
be connected to a PSE or PD through a non-MDI connector that is not isolated from the MDI leads 
needs to provide isolation between all accessible external conductors, including frame ground (if any), and the non-MDI connector. Accessible external conductors are specified in subclause 6.2.1b) of IEC60950-1:2001.

> This electrical isolation shall withstand at least one of the following electrical strength tests: 
a) 1500Vrms at 50Hz to 60Hz for 60s, applied as specified in subclause5.2.2 of IEC60950-1:2001.
b) 2250Vdc for 60s, applied as specified in subclause5.2.2 of IEC60950-1:2001.
c) An impulse test consisting of a 1500V, 10/700μs waveform, applied 10 times, with a 60s interval 
between pulses. The shape of the impulses shall be 10/700μs (10μs virtual front time, 700μsvirtual 
time of half value), as defined in IEC60950-1:2001 Annex N.

> There shall be no insulation breakdown, as defined in subclause5.2.2 of IEC60950-1:2001, during the test. The resistance after the test shall be at least 2MΩ, measured at 500Vdc. Conductive link segments that have differing isolation and grounding requirements shall have those requirements provided by the port-to-port isolation of network interface devices (NID).

[Source](https://www.silvertel.com/images/appsnotes/ANX-POE-Isolation-Barrier-explained.pdf)

Also, I saw your blog post today about the new [ESP32-POE-ISO](https://olimex.wordpress.com/2019/02/20/open-source-hardware-esp32-poe-board-now-has-new-version-with-3000vdc-galvanic-insulation) with 3000VDC isolation!

TsvetanUsunov, 2019-02-21:
Note that ESP32-POE is neither PD neither PSE and has no MDI. 
No one is supposed to touch this board while it's powered, and usually it is mounted in well isolated plastic box somewhere in the field.
If one wants to implement IP phone, router, or other device or want to use ESP32-POE as power supply for other equipment definitely should take measures to not allow people to make physical contact with the board metal parts which is the purpose of this part of the standard and the insulation may not be done necessary INSIDE the board, but with the enclosure where this board is installed.

jcstaudt, 2019-02-21:
I see. Based solely on this information, I understand that compliance for this spec is determined at the system level and not necessarily at the level of each individual component. If I'm correct, this would mean that the system (including ESP32-POE) could be compliant with this portion of the specification if isolation is also incorporated within the neighboring components (in an enclosure or otherwise). My interpretation of the snippet above tells me that power isolation is not required without onboard PD/PSE/MDI, which resolves this issue.

closed 2019-02-21 by jcstaudt

mlopezegi, 2019-10-10:
ESP32-POE is indeed a PD (Powered device) and has MDI (Medium Dependent Interface, in 100BaseT case: RJ45 and circuitry).
All external connectors and the frame must be isolated from line ethernet cable.
If there are no connectors available in the device and the frame is isolated from the circuits there is no problem.
Otherwise, ESP32-POE is not applicable and ESP32-POE-ISO must be used.