OLIMEX/ESP32-POE #4: Analog pins?

rdehuyss, 2019-04-08:
On the ESP32 POE, which are the analog pins? They are not mentioned in the connector on your website. 

Am I correct to assume that GPIO pins 39, 36, 35 and 34 can be used for measuring analog voltages?

d3v1c3nv11, 2019-04-09:
Nope. Only GPIO35, GPIO39 (EXT2 p.3, p.1)

tusharvb19, 2019-04-13:
So on GPIO35 which is HSPI and GPIO39 we can connect analog signal to measure???

danalytixx, 2022-07-27:
I am using an current sensor, to measure current. I used GPIO 39 and it is not working properly. Which pins should i use for this example? the other pins like 39 etc. are measuring voltage as i understood.

DanKoloff, 2022-07-27:
If by "analog pins" you mean ADC, it is like 12 pins that can be used for ADC, basically all pins that have the ADC function can be used, use this picture as a reference:

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/resources/ESP32-POE-GPIO.png

It is another question whether all of these 12 pins are free to use, some of them are probably used for one of their functions.