OLIMEX/ESP32-POE #6: MQTT client over the wired connection

zorce, 2019-07-15:
Hi,
I have tried most of the MQTT libraries without success to get it to work on the wired connection. Could you please provide an example on native MQTT and not azure example.

Best regards!

DanKoloff, 2019-07-16:
What about this: 

https://github.com/knolleary/pubsubclient

https://esp32.com/viewtopic.php?t=10328

zorce, 2019-07-16:
Thank you!
I finally managed to get it working! Thanks for the links!

DanKoloff, 2019-07-16:
Since we don't have a published example, if you can share your experience or demo code that would be awesome. I might even test it and publish it to this github if I manage to get something working.

zorce, 2019-07-16:
Will do. Currently testing stability. Will post my example here soon.

zorce, 2019-07-16:
Hi,
I have uploaded my example code at https://github.com/zorce/ESP32-POE_MQTT_example
Take a look and feel free to add it to your example repository.

lancelon, 2020-07-24:
Can I just confirm MQTT on the ESP32-POE is no problem? I'm about to buy a few for this express purpose and came across this thread...! Cheers

zorce, 2020-07-24:
For my purpose it has been stable enough. Have been running with 3 boards
for quite a while now.

DanKoloff, 2021-04-01:
We also tested Azure via Ethernet in Arduino IDE with this code, refer to the README.txt for installation instructions

[MQTT-Azure_ETH_WiFi.txt](https://github.com/OLIMEX/ESP32-POE/files/6241968/MQTT-Azure_ETH_WiFi.txt)
[README.txt](https://github.com/OLIMEX/ESP32-POE/files/6241969/README.txt)