OLIMEX/ESP32-POE #8: ESP32_PoE_Ethernet_IDFv4.0 Does not Build with v4.0

drdelgado, 2019-12-31:
Firstly I start:

```bash
butts@Ubuntu-precision-7530:~/esp/esp-idf$ git checkout release/v4.0 
Already on 'release/v4.0'
Your branch is up to date with 'origin/release/v4.0'.
butts@Ubuntu-precision-7530:~/esp/esp-idf$ git submodule update --init --recursive
```

However, your example code does not build here is my sdkconfig file:

### sdkconfig
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_TARGET_ESP32=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_SDK_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_SDK_PYTHON="python"
CONFIG_SDK_MAKE_WARN_UNDEFINED_VARIABLES=y
CONFIG_APP_COMPILE_TIME_DATE=y
# CONFIG_APP_EXCLUDE_PROJECT_VER_VAR is not set
# CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_NONE is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_ERROR is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_WARN is not set
CONFIG_BOOTLOADER_LOG_LEVEL_INFO=y
# CONFIG_BOOTLOADER_LOG_LEVEL_DEBUG is not set
# CONFIG_BOOTLOADER_LOG_LEVEL_VERBOSE is not set
CONFIG_BOOTLOADER_LOG_LEVEL=3
# CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_8V is not set
CONFIG_BOOTLOADER_VDDSDIO_BOOST_1_9V=y
# CONFIG_BOOTLOADER_FACTORY_RESET is not set
# CONFIG_BOOTLOADER_APP_TEST is not set
CONFIG_BOOTLOADER_WDT_ENABLE=y
# CONFIG_BOOTLOADER_WDT_DISABLE_IN_USER_CODE is not set
CONFIG_BOOTLOADER_WDT_TIME_MS=9000
# CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE is not set
# CONFIG_SECURE_SIGNED_APPS_NO_SECURE_BOOT is not set
# CONFIG_SECURE_BOOT_ENABLED is not set
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set
CONFIG_ESPTOOLPY_PORT="/dev/ttyUSB0"
CONFIG_ESPTOOLPY_BAUD_115200B=y
# CONFIG_ESPTOOLPY_BAUD_230400B is not set
# CONFIG_ESPTOOLPY_BAUD_921600B is not set
# CONFIG_ESPTOOLPY_BAUD_2MB is not set
# CONFIG_ESPTOOLPY_BAUD_OTHER is not set
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=115200
CONFIG_ESPTOOLPY_BAUD=115200
CONFIG_ESPTOOLPY_COMPRESSED=y
# CONFIG_ESPTOOLPY_FLASHMODE_QIO is not set
# CONFIG_ESPTOOLPY_FLASHMODE_QOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE_DIO=y
# CONFIG_ESPTOOLPY_FLASHMODE_DOUT is not set
CONFIG_ESPTOOLPY_FLASHMODE="dio"
# CONFIG_ESPTOOLPY_FLASHFREQ_80M is not set
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
# CONFIG_ESPTOOLPY_FLASHFREQ_26M is not set
# CONFIG_ESPTOOLPY_FLASHFREQ_20M is not set
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
# CONFIG_ESPTOOLPY_FLASHSIZE_1MB is not set
CONFIG_ESPTOOLPY_FLASHSIZE_2MB=y
# CONFIG_ESPTOOLPY_FLASHSIZE_4MB is not set
[... 700 lines cut ...]
In file included from /home/butts/esp/esp-idf/components/esp_eth/src/esp_eth.c:15:0:
/home/butts/esp/esp-idf/components/esp_eth/src/esp_eth.c: In function 'esp_eth_decrease_reference':
/home/butts/esp/esp-idf/components/newlib/include/stdatomic.h:279:30: error: request for member '__val' in something not a structure or union
  __atomic_fetch_sub(&(object)->__val, operand, order)
                              ^
/home/butts/esp/esp-idf/components/newlib/include/stdatomic.h:361:2: note: in expansion of macro 'atomic_fetch_sub_explicit'
  atomic_fetch_sub_explicit(object, operand, memory_order_seq_cst)
  ^
/home/butts/esp/esp-idf/components/esp_eth/src/esp_eth.c:352:5: note: in expansion of macro 'atomic_fetch_sub'
     atomic_fetch_sub(&eth_driver->ref_count, 1);
     ^
/home/butts/esp/esp-idf/components/esp_eth/src/esp_eth.c:352:5: error: incompatible type for argument 1 of '__atomic_fetch_sub'
/home/butts/esp/esp-idf/components/esp_eth/src/esp_eth.c: At top level:
cc1: warning: unrecognized command line option '-Wno-frame-address'
/home/butts/esp/esp-idf/make/component_wrapper.mk:291: recipe for target 'src/esp_eth.o' failed
make[1]: *** [src/esp_eth.o] Error 1
make[1]: *** Waiting for unfinished jobs....
CC build/esp32/cache_err_int.o
CC build/driver/pcnt.o
CC build/esp32/intr_alloc.o
/home/butts/esp/esp-idf/make/project.mk:590: recipe for target 'component-esp_eth-build' failed
make: *** [component-esp_eth-build] Error 2
```

Whats going on?

nick-atmosphereiot, 2020-01-12:
I'm getting the same error

DanKoloff, 2020-01-13:
This is somehow expected. We never tested with ESP-IDF newer than v3.3.1. The reason is simple - there is no stable tag for newer versions only "beta" and "dev" releases. 
[image]

HudsonProdigy, 2020-01-22:
I am also having this same error when trying to update from v3.3(LTS) to 4.0 rc or 3.3.1. was this ever resolved?

stiebrs, 2020-03-03:
Same thing, also while upgrading from 3.3

Stanimir-Petev, 2020-03-11:
There is an update of the example with the stable ESP-IDF v4.0.
Make sure you follow the instruction and install the version shown on this guide: https://docs.espressif.com/projects/esp-idf/en/stable/get-started/

Because last example was made with "older" 4.0 and that created the whole problem. There were few different versions all labeled as 4.0 but different between each other.

Try it and send your feedback if there are still issues.

bluelu, 2020-05-22:
Here is a 4.0.0 example which works for me:

`#include <stdio.h>
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "tcpip_adapter.h"
#include "esp_eth.h"
#include "esp_event.h"
#include "esp_log.h"
#include "driver/gpio.h"
#include "sdkconfig.h"

static const char *TAG = "Main";

/** Event handler for Ethernet events */
static void eth_event_handler(void *arg, esp_event_base_t event_base,
                              int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

    switch (event_id) {
    case ETHERNET_EVENT_CONNECTED:
        esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
        ESP_LOGI(TAG, "Ethernet Link Up");
        ESP_LOGI(TAG, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                 mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
        break;
    case ETHERNET_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "Ethernet Link Down");
        break;
    case ETHERNET_EVENT_START:
        ESP_LOGI(TAG, "Ethernet Started");
        break;
    case ETHERNET_EVENT_STOP:
        ESP_LOGI(TAG, "Ethernet Stopped");
        break;
    default:
        break;
    }
}

/** Event handler for IP_EVENT_ETH_GOT_IP */
static void got_ip_event_handler(void *arg, esp_event_base_t event_base,
                                 int32_t event_id, void *event_data)
{
    ip_event_got_ip_t *event = (ip_event_got_ip_t *) event_data;
    const tcpip_adapter_ip_info_t *ip_info = &event->ip_info;

    ESP_LOGI(TAG, "Ethernet Got IP Address");
    ESP_LOGI(TAG, "~~~~~~~~~~~");
    ESP_LOGI(TAG, "ETHIP:" IPSTR, IP2STR(&ip_info->ip));
    ESP_LOGI(TAG, "ETHMASK:" IPSTR, IP2STR(&ip_info->netmask));
    ESP_LOGI(TAG, "ETHGW:" IPSTR, IP2STR(&ip_info->gw));
    ESP_LOGI(TAG, "~~~~~~~~~~~");
}

#define CONFIG_EXAMPLE_ETH_PHY_RST_GPIO -1
#define CONFIG_EXAMPLE_ETH_PHY_ADDR 0
#define	PIN_PHY_POWER GPIO_NUM_12

void app_main()
{
    tcpip_adapter_init();

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    ESP_ERROR_CHECK(tcpip_adapter_set_default_eth_handlers());
    ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
    ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));

    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    phy_config.phy_addr = CONFIG_EXAMPLE_ETH_PHY_ADDR;
    phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;
    gpio_pad_select_gpio(PIN_PHY_POWER);
    gpio_set_direction(PIN_PHY_POWER,GPIO_MODE_OUTPUT);
    gpio_set_level(PIN_PHY_POWER, 1);
    vTaskDelay(pdMS_TO_TICKS(10));

    esp_eth_phy_t *phy = esp_eth_phy_new_lan8720(&phy_config);
    esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);

    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
    esp_eth_handle_t eth_handle = NULL;
    ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));
    ESP_ERROR_CHECK(esp_eth_start(eth_handle));
}

`

Stanimir-Petev, 2020-05-26:
> Here is a 4.0.0 example which works for me:
> 
> https://pastebin.com/8uy1KLbt

Can you upload the whole project (not only the main.c file)?
This means files like CMakeLists.txt, makefile, sdkconfig etc (check the full structure of the project). Because the macro values defined in sdkconfig are determined from the menuconfig and affect the compilation procress. So the same main.c source file may give different results depending on these values.

bluelu, 2020-05-26:
Here is the sdkconfig. I overwrote though some definitions directly in the c file.

#define CONFIG_EXAMPLE_ETH_PHY_RST_GPIO -1
#define CONFIG_EXAMPLE_ETH_PHY_ADDR 0
#define PIN_PHY_POWER GPIO_NUM_12

https://pastebin.com/SVyccrAq

Stanimir-Petev, 2020-05-27:
Unfortunately I still can't compile the code with the files you linked when I overwrite the ones from our project. I get this error:

`C:\esp-idf\esp-idf_v4.0\SPP\ESP32_PoE_Ethernet_IDFv4.0>idf.py flash
Checking Python dependencies...
Python requirements from C:\esp-idf\esp-idf_v4.0\requirements.txt are satisfied.
Adding flash's dependency "all" to list of actions
Executing action: all (aliases: build)
Running ninja in directory c:\esp-idf\esp-idf_v4.0\spp\esp32_poe_ethernet_idfv4.0\build
Executing "ninja all"...
[351/914] Generating ../../partition_table/partition-table.bin
FAILED: partition_table/partition-table.bin
cmd.exe /C "cd /D C:\esp-idf\esp-idf_v4.0\SPP\ESP32_PoE_Ethernet_IDFv4.0\build\esp-idf\partition_table && C:\Users\Stanimir5F\.espressif\python_env\idf4.0_py3.7_env\Scripts\python.exe C:/esp-idf/esp-idf_v4.0/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB C:/esp-idf/esp-idf_v4.0/components/partition_table/partitions_two_ota.csv C:/esp-idf/esp-idf_v4.0/SPP/ESP32_PoE_Ethernet_IDFv4.0/build/partition_table/partition-table.bin"
Partitions defined in 'C:/esp-idf/esp-idf_v4.0/components/partition_table/partitions_two_ota.csv' occupy 3.1MB of flash (3211264 bytes) which does not fit in configured flash size 2MB. Change the flash size in menuconfig under the 'Serial Flasher Config' menu.
[353/914] Building C object esp-idf/app_update/CMakeFiles/__idf_app_update.dir/esp_ota_ops.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

Can you upload somewhere a zip (or any other format) of the whole project? Check the structure of the project in the code section.

bluelu, 2020-05-27:
Hi,

You should be able to build a working project from the .c file I pasted. Your code fails because of a wrongly configured flash size, not related to any network code.

DanKoloff, 2020-05-27:
In the link @bluelu pasted ( https://pastebin.com/SVyccrAq ), this is improper:

CONFIG_ESPTOOLPY_FLASHSIZE="2MB"

bluelu, 2020-06-12:
I did some more tests using esphome, scanning ble devices and sending them over mqtt (> 10 messages per second).

The board will run fine for a few seconds, then eventually the link will go down once and come up again. Then after a while the link will go down but not go up anymore (e.g. just for less than < 1 second). There is not much in the logs, except that the ethernet got disconnected.

I made sure to not trigger any more mqtt sends, when the link is down, but it did not help.

Has anybody run this board stable with a "log" of traffic?

bluelu, 2020-06-22:
Hi,

In case anyone has the same problem:

The issue was related to an incompability to the POE switch I was using (netgear f108p). It works now when I force the ethernet controller not to advertise 100Mbit and fix it to 10Mbit.

dirkx, 2022-09-28:
I need to add below patch to make things work again against main.

```
diff --git a/examples/common_components/protocol_examples_common/eth_connect.c b/examples/common_components/protocol_examples_common/eth_connect.c
index 0b7d8f4b1a..36b41b10fd 100644
--- a/examples/common_components/protocol_examples_common/eth_connect.c
+++ b/examples/common_components/protocol_examples_common/eth_connect.c
@@ -97,6 +97,16 @@ static esp_netif_t *eth_start(void)
     eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
     phy_config.phy_addr = CONFIG_EXAMPLE_ETH_PHY_ADDR;
     phy_config.reset_gpio_num = CONFIG_EXAMPLE_ETH_PHY_RST_GPIO;
+
+#define CONFIG_PIN_PHY_POWER 12
+#ifdef CONFIG_PIN_PHY_POWER
+#warning "Power complied in."
+    esp_rom_gpio_pad_select_gpio(CONFIG_PIN_PHY_POWER);
+    gpio_set_direction(CONFIG_PIN_PHY_POWER, GPIO_MODE_OUTPUT);
+    gpio_set_level(CONFIG_PIN_PHY_POWER, 1);
+    vTaskDelay(pdMS_TO_TICKS(10));
+#endif
+

```

closed 2026-01-19 by TsvetanUsunov