OLIMEX/ESP32-POE #9: ESP32-POE-ISO Hanging up

TeKiLLa1985, 2020-01-15:
hi i used this code 

`#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 12

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      delay(1000);
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      Serial.println("Hostname: esp32-ethernet");
      delay(1000);
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
    delay(1000);
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}`

and get this output 

`14:34:58.966 -> e⸮⸮⸮0Q0080⸮X⸮H⸮ETH Started
14:35:00.279 -> Hostname: esp32-ethernet
14:42:36.269 -> ETH Connected`

and nothing else happend with a other ESP32-POE-ISO the sketch runs fine i tested it on 2 ESP32-POE-ISO and 3 different networks

I hope some one have a Idea to solve this issue

DanKoloff, 2020-01-16:
Each board has the Ethernet interface tested here empirically after manufacturing. So it is likely some sort of software or configuration issue. Try this example:

https://github.com/espressif/arduino-esp32/blob/1.0.3/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino

And follow the instructions here:

https://www.olimex.com/Products/IoT/ESP32/_resources/Arudino-ESP32.txt

The last few times someone encountered similar issue it turned out to be some improper network configuration, unrelated to the the ESP32-POE-ISO. For example here: 

https://www.olimex.com/forum/index.php?topic=7158.0

DanKoloff, 2020-01-16:
What operating system are you using?

TeKiLLa1985, 2020-01-16:
Win 10 Pro with the last updates

DanKoloff, 2020-01-16:
If the board was working fine, and then stopped working without any changes to the hardware or software setup - then either the board or something in the setup broke somehow.

Did you also test with the ESP-IDF example? This would reduce the chances of a software issue.

TeKiLLa1985, 2020-01-16:
i will try it with the IDF

closed 2020-10-15 by DanKoloff