OLIMEX/ESP32-POE #11: first upload worked, second did not

karls0, 2020-03-15:
### Hardware:
Board:							Olimex ESP32-PoE
Core Installation version:			Core 4.2.1  Home 3.1.1
IDE name:						platform.io 
Flash Frequency:					default
PSRAM enabled:                                    don't know
Upload Speed:					tried default, now at 115200
Computer OS:                                         Linux Mint 19.3

### Description:
I recently bought an Olimex ESP32-PoE and an Olimex ESP32-EVB and uploaded a test-sketch to both. No problem with this.
Next day I wanted to try another sketch, but the upload failed with 
 "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header"
From my experience with ESP8266 I checked the supply, but I measure 5.1V at the board. The USB-cable is the same as yesterday, but nevertheless I tried another and it did not help.

I also tried different versions of pressing RST- and BUT-button - no help.

Since I could upload a sketchd yesterday, I really do not understand what could be wrong. Thank you for any tips to help me.
Regards, Karl

DanKoloff, 2020-03-16:
Make sure you are using same USB port as previously. Also try re-installing the drivers, since you use Linux use these instructions: https://github.com/OLIMEX/ch340-dkms - check if this improves the behavior.

I'd also recommend testing with another software tool, maybe some update broke the software. Try Arduino for ESP32: https://github.com/espressif/arduino-esp32 or ESP-IDF

Finally, make sure to also check and report the issue to the platform.io community, maybe it is a well known problem.

If the issue remains tell me if:

Both the PoE and the EVB can't be programmed second time or just the PoE?

karls0, 2020-03-16:
Hi Dan,
thank you for the quick reply and your tips.
First I installed the driver from the OLIMEX-repository, rebooted and tried again. 
Uploading to a  OLIMEX ESP32-PoE, Olimex ESP32-EVB and AZ-delivery ESP32 Dev did not work - uploading to a ESP8266 works. This makes me think, that it cannot be the serial-driver.

I then installed Arduino 1.8.12 and the ESP-boards too (following https://www.heise.de/ct/artikel/Arduino-IDE-installieren-und-fit-machen-fuer-ESP8266-und-ESP32-4130814.html ) with the same results. No uploading to ESP32-boards, but everything OK with a ESP8266.  

Since the probability that all 3 new ESP32-boards are defective is very low, I must have missed something basic :-((  I have also tried to use a lower upload speed of 115200 - same results. 
Any further tips?

TIA, Karl

DanKoloff, 2020-03-16:
I don't think it is three defective boards. It is something trivial, like Linux gave new COM port number for the devices, or some other peripheral used and blocked the COM port, or the driver associated changed and it stopped toggling reset, maybe wrong board configuration was selected. What is the chip in the ESP8266 that works? Is it CH340T? If not - the comparison in behavior is inconclusive, since it probably uses different driver. If the ESP8266 board has CH340T then it is less likely to be a driver issue but still not completely out of question.

I think it is a popular problem so make sure to check for what others might have suggested online.

Also share the test program that you used. I might run it on our boards to compare.

karls0, 2020-03-16:
Hi Dan,
I finally found a solution :-)
https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/ describes how to solve the problem. After reading all the comments I started with a 100nF capacitor between EN and GND. This did not work. ThenI tried the 10µF and connected a resistor (the commenter says 1k - I had 5k6 at hand) from EN to +3.3V. Now the upload works and the program starts after uploading.
Thanks for your help and your tips.
Karl

DanKoloff, 2020-03-17:
@karls0 Hello,

Thanks for the update!

Can you confirm all the three boards worked after this fix? Can you also tell me what are the hardware revisions of each of the three boards as printed on the board?

The reason I need that information is to test here and maybe make some hardware changes. Thanks!

karls0, 2020-03-17:
Hi Dan,
I am at it. Did some tests with the AZ-delivery ESP32-Dev and now I tried the ESP32-PoE. Yesterday this worked as I told you. Today I got a new Error. There is no /dev/ttyUSB0 when I connect one of the Olimex-boards (with ch340), but is there with tht AZ-delivery which uses a CP2102. So I tried to install the ch340-driver again (as you told me in your first post). Finally I found out that I had to remove the driver completely and build and install it again. Now /dev/ttyUSB0 is here again :-))
I am continuing my work now and send the results asap.
Karl

karls0, 2020-03-17:
Hi Dan
find enclosed my findings about ploading code to ESP32-boards. I hope this helps you.
The HW-revisions are:
OLIMEX ESP32-PoE  Rev.D
OLIMEX ESP32-EVB  Rev.F
AZ-delivery ESP32 Dev   (no revision on board, but was bought only one week ago)

[upload-check.zip](https://github.com/OLIMEX/ESP32-POE/files/4343544/upload-check.zip)

hth, Karl

DanKoloff, 2020-03-17:
Thank you for the tests and the thorough reports! Pretty helpful. Future hardware revisions would have improved values on EN line but we are still deciding on what values exactly. Unfortunately, we can't replicate the "no upload scenario" here. That is why I have one final request - can you test also with 1uF if you have 1uF capacitor?

karls0, 2020-03-18:
Hi Dan, sorry for the delay. I did my tests with 1µF too and rearranged the table a bit. looks like 1µF is to small. 10M and a resistor between 4k7 and 10k seem to be best (for me). Did you think of a power-manager like TPS3839? Maybe this is a solution?

[upload-check.zip](https://github.com/OLIMEX/ESP32-POE/files/4348592/upload-check.zip)

Again, I hope my findings help. Please inform me of your solution.
Best wishes, Karl

karls0, 2020-03-20:
Hi again,
I forgot to mention that with some combinations reset did not work! Since I don't have a TPS3839DBRZ or a KA75290 for 3.3V I cannot test this, but as soon as I place my next order in China, I will order some of these and test again.
Best wishes and stay healthy!
Karl

karls0, 2020-03-23:
Hi,
as mentioned above, I think that a power-monitor could solve the problem. So I ordered a few TPS3809K33 (SMD) and a few MCP102-300 (for breadboard tests). Since the MCP102 has an open collector output, you can connect it directly to ESP_EN (pin4 on EXTERN1). But since there is button (RST) to GND, I suggest a resistor (I use 4k7 since it only draws current while RST is pressed) in between the output of the IC and EN.  
For me this is the best solution, since it is not dependent on the time constant and now even the RST-button works flawlessly. Now I can upload code and use the RST-button.
Regards, Karl