OLIMEX/ESP32-POE #13: Timed out waiting for PHY register

Gnubsch, 2020-04-15:
Hello,
I have bought three ESP32-POE-ISO boards With two copies the demo for the ethernet interface works. With the third board I get the timeout error. 
It has probably something to do with the clock. 

I used the ESP32_PoE_Ethernet_Arduino file for this.
Any tips how I can solve the problem?
I have already tried it on different PC's with different USB cables and ports.

Thanks a lot.

Error: Timed out waiting for PHY register 0x2 to have value 0x0007(mask 0xffff). Current value 0x0000[0m

Arduino 1.8.12 / Board-Lib esp32 1.0.4

DanKoloff, 2020-04-15:
What operating system are you using? What is the baud rate selected?

Gnubsch, 2020-04-15:
OS: Windows 10 Pro / 10.0.18363
Board: Olimex ESP32-POE-ISO Rev.C
Uploadspeed: 115200
Flash Frequency: 80 Mhz
Partition Scheme: Standard
Programmer: AVRSIP mkII
Driver: CH341Ser.INF 01/30/2019, 3.5.2019

`esptool.py v2.6
Serial port COM3
Connecting.....
Chip is ESP32D0WDQ5 (revision 1)
Features: WiFi, BT, Dual Core, VRef calibration in efuse, Coding Scheme None
MAC: 24:0a:c4:25:99:2c
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 6553.6 kbit/s)...
Hash of data verified.
Compressed 15856 bytes to 10276...

Writing at 0x00001000... (100 %)
Wrote 15856 bytes (10276 compressed) at 0x00001000 in 0.9 seconds (effective 139.7 kbit/s)...
Hash of data verified.
Compressed 706784 bytes to 393190...

Writing at 0x00010000... (4 %)
Writing at 0x00014000... (8 %)
Writing at 0x00018000... (12 %)
Writing at 0x0001c000... (16 %)
Writing at 0x00020000... (20 %)
Writing at 0x00024000... (25 %)
Writing at 0x00028000... (29 %)
Writing at 0x0002c000... (33 %)
Writing at 0x00030000... (37 %)
Writing at 0x00034000... (41 %)
Writing at 0x00038000... (45 %)
Writing at 0x0003c000... (50 %)
Writing at 0x00040000... (54 %)
Writing at 0x00044000... (58 %)
Writing at 0x00048000... (62 %)
Writing at 0x0004c000... (66 %)
Writing at 0x00050000... (70 %)
Writing at 0x00054000... (75 %)
Writing at 0x00058000... (79 %)
Writing at 0x0005c000... (83 %)
Writing at 0x00060000... (87 %)
Writing at 0x00064000... (91 %)
Writing at 0x00068000... (95 %)
Writing at 0x0006c000... (100 %)
Wrote 706784 bytes (393190 compressed) at 0x00010000 in 34.7 seconds (effective 163.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.0 seconds (effective 1536.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...`

DanKoloff, 2020-04-15:
You used this example, right:

https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino

Gnubsch, 2020-04-15:
Correct.

DanKoloff, 2020-04-15:
When you get the error, what happens if you press the on-board reset button?

Gnubsch, 2020-04-15:
The error occurs after restarting the device or when I press the hardware button. The same error occurs.

DanKoloff, 2020-04-16:
Please drop an e-mail at support@olimex.com with short introduction and link to this GitHub issue, I will tell you how to proceed with the return.

We don't have this problem here and each board's Ethernet is tested empirically so I have no idea what might be wrong. The clock comes from the ESP32, so I don't think it is clock issue.

Gnubsch, 2020-04-16:
Thank you for your efforts. I am writing the email to the support.

danieltang3434, 2020-10-12:
Hi @Gnubsch, did you get any feedback from Olimex? I'm facing the same problem also.

DanKoloff, 2026-09-24:
Closing this issue, it is stale and board design had gone multiple iterations in these 5 years.

closed 2026-09-24 by DanKoloff