OLIMEX/ESP32-POE #14: SD card reader conflicts with Ethernet interface PHY_PWR pin

cyberman54, 2020-05-21:
GPIO12 is used for PHY_PWR switch on this board.
But in ESP IDF it is used for Data3 of 4bit SDMMC interface.
Unfortunately forcing to 1bit SDMMC interface seems not to work with arduino esp32.
I posted a PR to solve this: https://github.com/espressif/arduino-esp32/pull/4020

DanKoloff, 2020-05-22:
Ummm 1bit SDMMC works with Arduino IDE, we have also defined it in our board configuration as "#define BOARD_HAS_1BIT_SDMMC":

https://github.com/espressif/arduino-esp32/blob/master/variants/esp32-poe/pins_arduino.h

Make sure you are selecting the proper board. Try this example:

https://raw.githubusercontent.com/OLIMEX/ESP32-POE/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_SD_Card_Arduino/ESP32_PoE_Ethernet_SD_Card_Arduino.ino

DanKoloff, 2020-05-22:
https://github.com/espressif/arduino-esp32/issues/3903

DanKoloff, 2020-05-22:
Thanks, I see it was merged now.

closed 2020-05-22 by DanKoloff

cyberman54, 2020-05-24:
For reference, from esp32 Forum:

https://www.esp32.com/viewtopic.php?t=4838