OLIMEX/ESP32-POE #19: SD card and Serial1 not working together

aboronkai, 2020-12-06:
I use Serial0 and Serial1 UARTs the same time, but when I try to use the SD card (with the given examples) the Serial1 stops working (Serial0 still OK),

What could be the problem?

Even if I don't do anything (read, write...) with the SD card, just call the SD_MMC.begin at the Setup(), the Serial1 doesn't respond.

aboronkai, 2020-12-10:
I made a ton of search on this topic, but found no usable answer.
BUT found the solution.
Interestingly if I placed the SD_MMC.begin() BEFORE the ETH.begin() line, the SD card was working, the serial1 was working.
I don't really understand why, but it works.

DanKoloff, 2020-12-15:
Did you update everything first? What you mentioned should not happen anymore if you use latest releases.

This used to happen because of this: https://github.com/espressif/arduino-esp32/issues/3903

The issue was re-iterated and clarified here: https://github.com/espressif/arduino-esp32/pull/4020

It was fixed in this commit: https://github.com/espressif/arduino-esp32/commit/594ee6d249f726a0f3335533b911a401d4eb028d

aboronkai, 2020-12-27:
Thanks for the reply!
I use Arduino 1.8.13, installed at 2020.oct. 13.
Using the default SD-MMC example from Olimex, the working mode is 1 bit by default. 
Anyway as I mentioned I found a way to work...

closed 2021-01-04 by DanKoloff