OLIMEX/ESP32-POE #20: Example uses depricated calls, please post example with 4.1++

bbulkow, 2021-01-06:
Given that Espressif has moved to ESP-IDF quite a while ago - maybe a year - perhaps it's time to stop using the depricated called, and post an example after the changes?

Here is the migration webpage:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/tcpip_adapter_migration.html

Here is the deprication example, exactly as one expects from 4.0 code in a post-4.2 world.

../main/ethernet_example_main.c: In function 'app_main':
../main/ethernet_example_main.c:69:5: warning: 'tcpip_adapter_init' is deprecated [-Wdeprecated-declarations]
     tcpip_adapter_init();

DanKoloff, 2021-01-06:
Thanks for the help. It is in our TODO list but our software back log is huge and we have trouble updating everything all the time. Overall, our goal had never been to provide the best examples with the latest software tools, just good enough examples to prove that the hardware works. There is a reason our projects are published here on the GitHub - we hope that more knowledgeable people can contribute with their feedback or with patches.

bbulkow, 2021-01-06:
If it were true that these changes were in the last few months, I would have some sympathy. However, I believe the 4.1 release is somewhere around a year old, and contains very major changes to the networking stack. If you aren't able to push the example in this period of time, I would suggest updating the readme to state you only support 4.0 and nothing more recent.

It is regrettably the Espressif people seem so flagrant about breaking shipping APIs, but they are, which makes it hard to just switch my project back to 4.0.

DanKoloff, 2021-01-06:
Is the tag one year old, tho?
[image]

The project folders of each of our demos has the exact version of ESP-IDF required mentioned in the folder name and also every resource is available before and without any purchase, so it clear what resources are exactly available. Also the examples, can be used as basis for any version of ESP-IDF. Most of the demos are simple pin assignments.

bbulkow, 2021-01-06:
It's much older than that. Espressif moves the tags.

closed 2021-01-06 by bbulkow

bbulkow, 2021-01-07:
https://github.com/bbulkow/Olimex-ESP32-POE-example

DanKoloff, 2021-01-07:
Thank for this, I tested what you published and it works here, one small thing that I spotted in the description is that the boards use CH340, not CH341 but the difference between them is not big anyway. My main problem is the initial report for the issue. I also tested our 4.0 version example and it compiles here with version 4.1 and 4.2, it also works when empirically tested, so I am puzzled what was the original issue? The 4.0 project didn't build at your side? Maybe I don't understand something.

DanKoloff, 2021-01-08:
We've added a 4.2 example, like the previous ones - we just used the default demo and did some re-mapping to fit ESP32-PoE's hardware. It is located here https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv4.2