OLIMEX/ESP32-POE #21: Can it work like an access point?

fr33soul, 2021-03-16:
Is there any way to make it work as an access point?
For example, does it take network data from the ethernet port and release it through the wifi with only one IP address for both? Even with mesh architecture?!

DanKoloff, 2021-03-17:
Sure both Ethernet and WIFI can work at the same time. This is a simple test for using both WIFI and Ethernet under Arduino IDE:

https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_WIFI_test

If you want more info search online for:

"esp32 ethernet wifi bridge"

fr33soul, 2021-03-17:
I tried to google and the only solution would look like this: [https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap](url)
now i tried to compare your eth_wifi_test with that of espressif. I would like to try with Espressif's ESP-IDF solution and I wanted to know if Are there any parameters that are specific to your Olimex that I need to take into consideration?

fr33soul, 2021-03-17:
ho notato che anche voi avete ESP-IDF [https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv4.2](url)
si può utilizzare per questo scopo? a cosa si riferisce il vostro esempio ESP-IDF?

DanKoloff, 2021-03-17:
We have ESP-IDF examples and you can use the example that you linked, of course, as basis for Ethernet connectivity. But we haven't published ESP-IDF example that demonstrates both WIFI and Ethernet (in the same example).

fr33soul, 2021-03-17:
ok i will try the Espressif example than. I will let you know if it works.

closed 2021-05-17 by DanKoloff