OLIMEX/ESP32-POE #24: Ethernet + DMA I/O driven SPI display causes display corruption.

jonshouse1, 2021-08-02:
I am having trouble using a display with the POE board, issue described here:

https://github.com/espressif/esp-iot-solution/issues/110

DanKoloff, 2021-08-17:
1) Looking at your video you have ESP32-POE-ISO. Not ESP32-POE...

2) A lot of cables connected at the header, probably you are using a pin that is already used for on-board peripheral. Quite likely one of the pins related to the Ethernet. Double check if the pins you are using are free.

3) What is the display that you are using? Is it ours MOD-LCD2.8RTP, it seems so? Why are you not connecting it to the board via the UEXT cable?

jonshouse1, 2021-08-17:
> Looking at your video you have ESP32-POE-ISO. Not ESP32-POE...

I have both the ISO and non ISO models, I get the same results on both.

>  A lot of cables connected at the header, probably you are using a pin that is already used for on-board peripheral. Quite likely one of the pins related to the Ethernet. Double check if the pins you are using are free.

https://github.com/jonshouse1/ESPVNCC/blob/main/esp32_lcd_touch_screen_wiring.png
The only used connection has button with a pullup. I tried moving my GPIOs around, it did not make a difference.

>   What is the display that you are using? Is it ours MOD-LCD2.8RTP, it seems so? Why are you not connecting it to the board via the UEXT cable?

Not your display module, no. I have several projects under way with this board (If I can make it reliable!), I am using the header to connect sensors rather than display.

DanKoloff, 2021-08-17:
Meanwhile I tested with our MOD-LCD2.8RTP and ESP32-POE-ISO with and without Ethernet connected and it has no such issue. Mind that our display uses same controller. Here is a picture of my setup:

[image]

Here is the code and libraries that I used (quick and dirty combination of LCD and Ethernet demo):

[lcd-esp32-poe-iso.zip](https://github.com/OLIMEX/ESP32-POE/files/6998136/lcd-esp32-poe-iso.zip)

DanKoloff, 2021-08-17:
The hardware connection that you linked seem alright on first look. 

If you have both ESP32-POE and ESP32-POE-ISO, start debugging the issue by using ESP32-POE. Leave ESP32-POE-ISO aside. The reason is that ESP32-POE-ISO has less free pins.

jonshouse1, 2021-08-17:
I am not a C++ programmer or Arduino ide user.

The best test I have found is to write a single colour to every line of the display over and over again as fast as possible, then send Ethernet data to the host and see if the display corrupts. Because the display is a single colour only your eye is drawn instantly to any change in colour.

At the moment it looks like the ESP IDF SPI driver has a bug that I am finding but it is a little early to be completely sure.