OLIMEX/ESP32-POE #30: does not build, missing files

gherlein, 2022-05-08:
https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ESP-IDF/README.md#common-configurations 404s

Also the build fails:

../main/ethernet_example_main.c:13:10: fatal error: tcpip_adapter.h: No such file or directory
 #include "tcpip_adapter.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.

I suspect I need to install or config something, but the [README](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ESP-IDF/README.md) is missing.

DanKoloff, 2022-05-13:
Well, if you wish to compile, for example "ESP32_PoE_Ethernet_IDFv4.4", it is recommended to use ESP-IDF version 4.4.

ESP-IDF v5.0-dev-2586-ga82e6e63d9 is nightly developer build for ESP-IDF developers. It is not an official release aimed at ESP32 developers, but aimed at ESP-IDF developers.

gherlein, 2022-05-15:
Good catch, but that's not the problem.  I did revert to using 4.4.  The OLIMEX repo is copied from the master Espressif repo but uses a relative link. The broken spot is "See common pin assignments for Ethernet examples from [upper level](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ESP-IDF/README.md#common-pin-assignments)."

The proper link is:  https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ESP-IDF/ESP32_PoE_Ethernet_IDFv4.0

I'll submit a PR.

I can now build this and it works.  But I am trying to understand the deltas between this working example and the official sample app.  It looks like the working app has:

 Ethernet Type (Internal EMAC)  --->
    Ethernet PHY Device (LAN8720)  --->
(23) SMI MDC GPIO number
(18) SMI MDIO GPIO number
(-1) PHY Reset GPIO number
(0) PHY Address

Closest I can get with the official ethernet sample app is:

[*] Internal EMAC
        Ethernet PHY Device (LAN87xx)  --->
(23)    SMI MDC GPIO number (NEW)
(18)    SMI MDIO GPIO number (NEW)
(-1)    PHY Reset GPIO number
(0)     PHY Address

but that crashes:

0x400810c4: call_start_cpu1 at /home/gherlein/esp/esp-idf/components/esp_system/port/cpu_start.c:160

I (242) cpu_start: App cpu up.
I (269) cpu_start: Pro cpu start user code
I (269) cpu_start: cpu freq: 160000000
I (269) cpu_start: Application information:
I (274) cpu_start: Project name:     ethernet_basic
I (279) cpu_start: App version:      1
I (284) cpu_start: Compile time:     May 15 2022 06:53:02
I (290) cpu_start: ELF file SHA256:  305b8665e38598d2...
I (296) cpu_start: ESP-IDF:          v4.4.1-dirty
I (301) heap_init: Initializing. RAM available for dynamic allocation:
I (308) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (314) heap_init: At 3FFB3BF0 len 0002C410 (177 KiB): DRAM
I (321) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (327) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (333) heap_init: At 4008B9CC len 00014634 (81 KiB): IRAM
I (341) spi_flash: detected chip: generic
I (344) spi_flash: flash io: dio
I (349) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (459) esp.emac: emac_esp32_init(349): reset timeout
E (459) esp_eth: esp_eth_driver_install(214): init mac failed
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x40085548
0x40085548: _esp_error_check_failed at /home/gherlein/esp/esp-idf/components/esp_system/esp_err.c:42

file: "../main/ethernet_example_main.c" line 122
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

abort() was called at PC 0x4008554b on core 0
0x4008554b: _esp_error_check_failed at /home/gherlein/esp/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x40081932:0x3ffb67a00x40085555:0x3ffb67c0 0x4008a6be:0x3ffb67e0 0x4008554b:0x3ffb6850 0x400d6359:0x3ffb6870 0x400ffdf9:0x3ffb6900 0x40088169:0x3ffb6920 
0x40081932: panic_abort at /home/gherlein/esp/esp-idf/components/esp_system/panic.c:402

0x40085555: esp_system_abort at /home/gherlein/esp/esp-idf/components/esp_system/esp_system.c:128

0x4008a6be: abort at /home/gherlein/esp/esp-idf/components/newlib/abort.c:46

0x4008554b: _esp_error_check_failed at /home/gherlein/esp/esp-idf/components/esp_system/esp_err.c:43

0x400d6359: app_main at /home/gherlein/src/esp32-poe-ea/basic/build/../main/ethernet_example_main.c:122 (discriminator 1)

0x400ffdf9: main_task at /home/gherlein/esp/esp-idf/components/freertos/port/port_common.c:129 (discriminator 2)

0x40088169: vPortTaskWrapper at /home/gherlein/esp/esp-idf/components/freertos/port/xtensa/port.c:131

gherlein, 2022-05-15:
Fixed a few other settings, but this one

CONFIG_EXAMPLE_ETH_PHY_LAN8720=y                              | CONFIG_EXAMPLE_ETH_PHY_LAN87XX=y

is probably causing the crash now.

I (382) system_api: read default base MAC address from EFUSE
E (402) lan87xx: lan87xx_init(499): wrong chip ID
E (402) esp_eth: esp_eth_driver_install(215): init phy failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40085548
0x40085548: _esp_error_check_failed at /home/gherlein/esp/esp-idf/components/esp_system/esp_err.c:42

file: "../main/ethernet_example_main.c" line 122
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

abort() was called at PC 0x4008554b on core 0
0x4008554b: _esp_error_check_failed at /home/gherlein/esp/esp-idf/components/esp_system/esp_err.c:43

I suspect I need to install the driver code, but with no instructions this is not easy.

gherlein, 2022-05-15:
oh hoo boy, the actual code is substantially different.

I guess we can close this issue.  The real problem is that to use this board on a new project my only choice is to basically derive off this sample code, preventing me from taking advantage of the nice "create a new project" features in Visual Code.  Boo.

closed 2022-07-20 by DanKoloff