OLIMEX/ESP32-POE #32: Example ESP32_PoE_Ethernet_Arduino does not connect (last event SYSTEM_EVENT_ROC_DONE ) ESP32-POE Rev G

edrethardo, 2022-06-30:
Heyho, 
I'm using [the example code ](https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino) with added debug prints on the ESP32-POE Rev G Board

My console ouput is:
starting Ethernet
ETH Any other event 18 
ETH Any other event 20
ETH Any other event 22

I'm connected to a Router with DHCP but I'm not using POE right now as I'm using the serial console via USB for debugging purposes to avoid damaging my Computer. The device is powered by my computers USB 3 connector 

it's notable that the gateway and dns server are not the same in our network, could it be I need to set those parameters manually?
Did maybe anything relevant to this change between Rev.B and G I didn't notice?
do I explicitly need to use POE in this example?
My code is attached below

```
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 12

#include <ETH.h>

static bool eth_connected = false;

void WiFiEvent(WiFiEvent_t event)
{
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("esp32-ethernet");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
    Serial.printf("ETH Any other event %u \n", event);
      break;
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void setup()
{
  Serial.begin(115200);
  delay(10000);
  Serial.println("\n starting Ethernet");
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}

void loop()
{
  if (eth_connected) {
    testClient("google.com", 80);
  }
  delay(10000);
}
```

Thank You for any suggestions

FStefanni, 2022-07-19:
Hi,

I have a rev D, but it does not work too.
I am using PoE with 48V, since it is stated that the chip to work must be powered.

No fix found at the moment...

Regards

FStefanni, 2022-07-19:
Hi,

some news: it seems to be a BSP bug, that I reported here:

<https://github.com/espressif/arduino-esp32/issues/7007>

Regards.

FStefanni, 2022-07-20:
Hi,

the issue is resolved: as pointed out in the previously linked issue, the updated example works.
Events have changed names, but the Olimex example has not been updated.
Here the link to the updated official esp32/arduino example: <https://github.com/espressif/arduino-esp32/blob/master/libraries/Ethernet/examples/ETH_LAN8720/ETH_LAN8720.ino>

Regards

DanKoloff, 2022-07-20:
Thanks for the update, I will keep it open as a reminder until we find time to update all Ethernet examples to the new definitions.