OLIMEX/ESP32-POE #33: Can GPIO5 be used ? Reading 2,5V on it all the time

Ulrar, 2022-09-08:
Hi,

I've been using Panasonic PIR sensors connected to esp32 (not poe) boards without issues, configured with esphome's binary_sensor in input_pulldown mode.

Trying to use the same config on an esp32-poe however and I can't get it to work on GPIO5, that pin reads 2,5V regardless of what's connected to it. [This PNG](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/resources/ESP32-POE-GPIO.png) doesn't say anything about that pin, but is there something connected to it already that would prevent me from using it for a binary_sensor ?

Thanks

DanKoloff, 2022-09-09:
GPIO5 has a 10k external pull-up - check it the schematic near the UEXT part. I see in your code "input_pulldown" so it probably creates some sort of divider (pull-up - pull-down). I don't know what is the sensor and the requirements but the issue is probably related to the hardware pullup on the board. There are different ways to overcome that issue but that would depend on the exact hardware and goals. If it was me, I'd probably fix this issue by removing the pull-up resistor - unsolder R36.

Ulrar, 2022-09-09:
Okay, I went and tried a bunch of pins and after quite a few I found one that worked : GPIO32.
So you're saying some of the pins are connected to pullup resistors but it's not mentioned on that png on the website, is there another document I should be referring to when choosing pins ?
Thanks

DanKoloff, 2022-09-09:
It is always best idea to check the schematic, or the export of the schematic in PDF format:

https://github.com/OLIMEX/ESP32-POE/blob/master/HARDWARE/ESP32-PoE-hardware-revision-I/ESP32-PoE_Rev_I.pdf

Download it and find and trace the GPIO you are interested in. ESP32 chip doesn't have a lot of pins and there is heavy multiplexing usually, so finding absolutely free pin is tough. But usually the GPIO pins at the UEXT are free to use, as long as you are aware of the pull ups.

Ulrar, 2022-09-13:
Ah, that's very helpful, I see the resistor you were referring to for GPIO5 now, so I know what to look out for.
Thanks !

closed 2022-09-13 by Ulrar