OLIMEX/ESP32-POE #36: Question about WiFi and Ethernet running simultaneously

gonzabrusco, 2022-11-29:
Hello. I have one question about how this example works:

https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_WIFI_test/ESP32_PoE_Ethernet_WIFI_test.ino

How do you know or how do you force which connection is made using Ethernet and which one is made using WiFi? It not very clear because you always initialize a WiFiClient object but there's no hint of which connection will use to reach internet. 

Thanks
Gonzalo

DanKoloff, 2022-12-01:
No idea, we just combined the two examples and they worked. Might be better idea to ask in the espressif forums for ESP32 or in the Arduino or Arduino for ESP32 forums.

0x0fe, 2023-02-06:
clearly this example is not valid, they use the exact same wificlient for both connection with no action to ensure which one connects to which link, besides, IDF doesnt have any mechanism as of now to set a particular client to a particular link, so all this example does is keep connecting via the same link. in this case it must be via wifi probably since it is the last link which has been setup.