OLIMEX/ESP32-POE #37: ESP32-POE-WROVER and Ethernet

mikkle, 2023-05-30:
Hello,

Since the WROVER-edition of EP32-POE uses GPIOs 16 and 17 for the PSRAM, it would seem there is a conflict with the Ethernet PHY, which also depends on GPIO17.

By default, and in the sample code (Arduino framework) it says
`#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT`
With that setting, ETH.begin() fails.
I got the PHY to start up by overriding that to
`#define ETH_CLK_MODE ETH_CLOCK_GPIO0_OUT`
but that seems a bit random, since the schematic for ESP32-POE shows pin 5 (XTAL/CLKIN) as connected to GPIO17.

And, indeed, I have some code that runs fine on the WROOM-edition, but borks on the WROVER-edition when powered over ethernet (which basically means that the the Eth connection is up).

So the big question: Is it at all possible to use the wired ethernet on ESP32-POE-WROVER, and if there is, are there any other restrictions apart from not using GPIOs 16-17 compared to the WROOM-edition - any low-level config options/settings that can solve the (as I see it) internal pin conflict?

Cheers,
Mikkle

DanKoloff, 2023-05-31:
Good point. In WROVER version, GPIO0 serves as EMAC clock. Refer to the picture that is an excerpt from the schematic:

[image]

mikkle, 2023-05-31:
Thanks very much!

By that except of the schematic, is GPIO33 safe to use on the WROVER, or will it potentially interfere with the PSRAM SPI on GPIO16?

Cheers
Mikkle

DanKoloff, 2023-06-01:
Since the WROVER board has no GPIO16, we connect GPIO33 on its place. GPIO33 can be used for all purposes but it would be routed to the UEXT (this is done so when we use WROVER, there is no free pin on the UEXT).

mikkle, 2023-06-01:
Ah, thanks, got it, and just tested that indeed, that "GPIO16" on both UEXT and EXT2 are connected to and driven from GPIO33.
Thanks again!

closed 2023-06-01 by mikkle