OLIMEX/ESP32-POE #39: Do these boards use the ethernet for network too?

buzurk26, 2023-06-21:
Hi 

Looking at these boards for home automation and want hard wired where possible, I have a PoE switch already

Do these boards also communicate over the ethernet port to your network or are yous till required to joint he esp32 module to your WIFI?

Do these support running ESPHome??

DanKoloff, 2023-06-21:
> Do these boards also communicate over the ethernet port to your network or are yous till required to joint he esp32 module to your WIFI?

I don't understand this question. Please paraphrase. Are you asking of Ethernet and WIFI can work simultaneously? 

> Do these support running ESPHome??

A lot of customers have reported success in using ESP32-POE with esphome. Just Google "ESP32-POE esphome" and check the results for customer feedback.

woodenplastic, 2023-07-05:
Hi, when i try the examples i cant get a ethernet connection. Wifi works, POE aswell. There seeems to be an issue Dan. Can you please tell me how to give you information?

DanKoloff, 2023-07-06:
@woodenplastic Which Olimex ESP32 board exactly do you own? ESP32-POE?

woodenplastic, 2023-07-06:
ESP32-POE-ISO-EA-IND is the name, but i couldnt find the revision.

woodenplastic, 2023-07-06:
digging through some forums i found this solution:
when using the arduino IDE you have to change your example posted here.
all SYSTEM_EVENT_ETH... commands have to be changed to ARDUINO_EVENT_ETH....
then everything works. why that is i dont know. maybe you could make the examples clearer for beginners like me.

DanKoloff, 2023-07-06:
The ETH_LAN8720 example code from Arduino IDE works just fine with ESP32-POE-ISO board and should be used as starting point. No need of any changes. 

[image]

Make sure you have selected Olimex ESP32-POE-ISO from the board selector.

If the issue remains with same code and settings, then probably the issue is somewhere in your network equipment or settings (like bad LAN cable, no DHCP enabled, router filtering or blocking the board, etc).

B48D81EFCC, 2023-07-30:
> Do these support running ESPHome??

Hi,

I am using several ESP-POE and ESP-POE-ISO modules exclusively with esphome.
I use the RJ45 for the network connection (and power supply of course). I disabled Wifi.
The modules are running 24/7. No issues so far.

thermseekr, 2023-08-10:
Take a look at the Tasmota project here on Guthub. It is an open source firmware to the ESP8266 and ESP32. I think it will do basically everything you need regarding home automation. And there is a template to enable the ethernet port on the ESP32-POE and -POE-ISO.

https://github.com/arendst/Tasmota