OLIMEX/ESP32-POE #40: Private DNS Zone

podexperfectussum, 2023-06-28:
Hello,

I am testing the ESP32-POE-ISO using the [suggested ethernet example](https://www.olimex.com/Products/IoT/ESP32/_resources/Arudino-ESP32.txt).  As long as the "host" I send to testClient is a public FQDN (ends in .com, .org, or .net), it works.  If I use an internal lan FQDN (*.local, *.lan), it does not.  Is this expected behavior?

DanKoloff, 2023-06-28:
As you can see from the instructions, we just point people on how to install official ESP32 support for Arduino and use default Ethernet example. Consider asking the ESP32 Arduino community about that, maybe it is something well known. Also quick google returns:

https://forum.arduino.cc/t/arduino-local-doesnt-work/184371/19

https://forum.arduino.cc/t/not-able-to-connect-arduino-to-localhost/664556/2

podexperfectussum, 2023-06-28:
Thanks, I'll find somewhere to ask questions.

I saw those yesterday when I was using other search engines for a couple of hours. Neither of those is applicable here though, they're trying to find their Arduino on the network from other things where as I'm trying to find things on the network from the esp32-poe-iso running Arduino.  So, the opposite.

closed 2023-06-28 by podexperfectussum