OLIMEX/ESP32-POE #41: R-DET and UVLO voltage divider

thermseekr, 2023-08-08:
Hello,

First of all, thank you very much for making your design open source. It helped me a lot to understand how to design and build ethernet and PoE around the ESP32. I have sucessfully designed and built a board with ethernet and 16 inputs/16 outputs using MCP23017 expanders, and now I'm working on a version with PoE supply. I have a doubt about the resistor choice for R-DET on the TPS2376 that I would like to ask you about.

The IEEE 802.3af/at standards state that the allowed range for R-DET is 23,75k to 26,25k. For most applications a 24,9k resistor is recommended, and this is what we see on ESP32-POE.

Now considering the UVLO voltage divider: the voltage at the UVLO pin should be 2,5V when VDD is at the desired turn on voltage. On the ESP32-POE there is a 130k/10k divider. Calculating for 2,5V on the tap we find that the voltage at VDD will be 35V, which is OK considering the 42V maximum turn  on voltage and 30V minimum turn off voltages.

The problem is that according to the TPS datasheet, the total resistance of the voltage divider (140k) will appear in parallel with R-DET (24,9k), bringing the resulting value of R-DET down to 21,14k which is below the minimum allowed value. Is there any standard detail or design consideration that I'm missing?

Thanks
Tales

thermseekr, 2023-08-08:
If you check TI's SLVU188 Reference Design, it uses 909k/61.9k for the voltage divider at UVLO, and 25.5k at R-DET, which combined gives a 24847R resistor at R-DET. It makes sense.

DanKoloff, 2023-08-21:
Thanks for questions and the feedback. Was there any issue with the original values that you noticed or is it theoretical question?

I think the discrepancy in values came from the fact that we originally used TPS2375 but it was unavailable for purchase for some time and that resistor divider stayed the same when we purchased TPS2376.I believe we used values suggested in SLVU108.

We will look into the values for next hardware revision, but not sure whether we stock TPS2375 or TPS2376 (if we are back to TPS2375 the values would have to stay as they are). I am forwarding this to the hardware team.

thermseekr, 2023-08-21:
My boards are from an older revision and I probably have the TPS2375. So no issues here.

I upgraded a 16in/16out board that I use internally at work to include the ESP32 onboard and I used your design as reference for power and ethernet, that's how I noticed the discrepancy.

You can set the resistor values for UVLO divider and RDET in such a way that the TPS2376 will behave exactly like the TPS2375, compliant to the IEEE standards. The values are in the datasheet, I can't check them now but they're easy to find.

Thanks for replying! Cheers!

DanKoloff, 2023-10-17:
It seems like this issue was resolved, closing.

closed 2023-10-17 by DanKoloff