OLIMEX/ESP32-POE #43: Capacitors in revision L are too high

fredlcore, 2023-09-22:
Hi, 
my BSB-LAN board (https://github.com/fredlcore/BSB-LAN/blob/master/BSB_LAN/schematics/Logic%20Level%20Adapter.jpg) is no longer fitting nicely on a revision L board because the two capacitors right next to the UEXT connector are too high, see attached image.
Could you adjust this in the next revision because it fit perfectly fine before and I assume that other UEXT boards that plug right in will be affected, too?
Thanks
[image]

DanKoloff, 2023-09-25:
Which revision did you use previously? Not sure if it would be possible to alter the design, I suspect that these capacitors were introduced when we migrated to the Texas PoE chip. Previously it was possible to place them sideways but now there is no space left. As a workaround use a cable between the ESP32-PoE and your board or think of a spacer of two connectors (male-female) to place in-between.

fredlcore, 2023-09-25:
Thanks for getting back to me. I think the previous revision was "K". In which revision was the Texas PoE introduced? 
A cable would not be ideal because you'd then have to fixate the board in some other way. Now our project offers users the STL of a 3D printed case where everything just fits.
As for the spacer, do you have an image on how that would look like?
I assume that other projects will be affected by this as well, so maybe it's worth to look into it in terms of alternative board population or something like that because just plugging a daughter-board into the UEXT connector was always an advantage compared to other microcontrollers.

DanKoloff, 2023-09-25:
I think your older boards were older than revision G. I compared revisions older than G and newer ones and I understand the problem. Previously we bent the two capacitors so that they were placed side ways on top of the PCB. We don't do that anymore and keep them straight up. Both ways have their advantages and disadvantages. But advantages of having them up exceeded those of having them bent. I guess the advantage of having them sideways was the better access to the UEXT connector and the disadvantages were too many - harder to solder (had to be soldered on top); lack of uniformity between different boards (some are bent slightly right, other slightly left); possible reliability issues with the capacitor bent on top of a DCDC converter that heats up (electrolytic capacitors might dry up faster when placed on top of that hot convertor); it takes more time to manufacture.

We will have to discuss it again if we want bent capacitors back in a newer revision and when there is a decision I'd let you know.

fredlcore, 2023-09-25:
Thank you. I understand that this was most likely not an easy decision, but if you keep the accessibility of the UEXT connector in mind, I'm sure you will find the overall best solution. Thank you.

closed 2023-09-25 by fredlcore

DanKoloff, 2023-09-27:
Hello,

After reviewing this change the team design team decided to not revert this change. The capacitors would be placed the current way. Sorry for the inconvenience and compatibility issues caused, but reliability of the board had been deemed more important.

The ways to circumvent this change are:

- alter your design and use higher connector (there are taller 10-pin female connectors) compared to the one you used
- create a spacer from female and male connectors soldered back to back
- use a cable
- trim the size of the PCB or mill the edge (if possible)

Best regards,
Lub/OLIMEX

fredlcore, 2023-09-27:
Thank you for getting back to me. I can understand the reasons, but such changes are nevertheless disappointing when they happen without advance notive. If you have a significant amount of boards prepared where your customers have to tell you that they no longer fit, then that is not the most reliable choice for a microcontroller. Maybe something to keep in mind for future changes that break especially hardware compatibility.

DanKoloff, 2023-12-12:
I've forwarded the suggestion to the team and they concluded moving the UEXT would create even more compatibility issues for other customers that used it. Again sorry for this change but it was necessary for reliability and safety of the board and the products it gets embedded into.

fredlcore, 2023-12-12:
Ok, understood, thanks for checking.