OLIMEX/ESP32-POE #49: Power supply via POE not stable

fredlcore, 2024-03-28:
I'm maintaining a hardware project (https://www.bsb-lan.de) where probably half of the installations run on an Olimex (either EVB or POE). About a dozen users using the Olimex ESP32-POE have now reported back to me that when they use WiFi on the Olimex, the system becomes unstable which shows in disconnects or slow WiFi connections.
The matter immediately changes when powering the Olimex via a stable USB power supply (a laptop's USB for example) and the issues are gone. Also, the matter does not seem to be noticeable if an Ethernet connection instead of WiFi is used.

I have also created an issue in the Olimex EVB section, feel free to close one of them if that is better for your workflow.

For the time being, we will add a recommendation to only power the Olimex via a reliable USB power supply, but since POE powering is often a specific demand by our users, I wanted to let you know about this problem, so that hopefully you can fix it in the next hardware revision.

thermseekr, 2024-03-29:
This is very odd. I'll be following out of curiosity. I developed an ESP32 based system and I used Olimex ESP32-POE as one of the hardware references so I studied the circuit a lot. As far as I could see the PoE power supply is pretty standard, nothing weird in it. I haven't studied a lot the isolated version tho.

Also, I'm curious as for why someone would connect PoE and use WiFi. Since you already have an Ethernet cable connected, why not use it also for data? Is it only an analog 24/48V power connection? Or do you need two different IP addresses?

Cheers.
Tales.

fredlcore, 2024-03-29:
You're right, it doesn't really make sense to have WiFi running on the POE, and I would have to check if they actually did run the system on WiFi (which, in my project, makes sense because it sets up an AP so you can configure it through via WiFi). 
I mentioned WiFi because on the EVB this was the most obvious proof for the instability of the system, but it was not limited to that. On both boards, however, the problems ceased once it was powered by a reliable USB power supply. If the power supply was cheap (despite high ampere ratings), it could still result in the same instabilities as when powered via barrel connector or POE.

fredlcore, 2024-10-18:
Most likely the reason for this issue is the same as @DanKoloff explained [here](https://github.com/OLIMEX/ESP32-EVB/issues/57#issuecomment-2162757027).

fredlcore, 2025-03-12:
Just FYI for everyone who is getting here via Google: We came to a conclusion [in this thread](https://github.com/fredlcore/BSB-LAN/issues/715#issuecomment-2716701711) that the corrupted data can be taken care of if the data stream has some form of checksum in place, as at least 97% of the data transmitted in the test scenario is not affected. For those setups that rely on accurately receiving via GPIO36 (and maybe other GPIOs as well), these should not power their Olimex via PoE or barrel power plug.