OLIMEX/ESP32-POE #53: 3D File of the entire ESP-POE

bmueller78, 2024-10-29:
Could you please upload the 3D file in STEP format of the entire ESP-POE Rev. L? Thank you very much.

DanKoloff, 2024-10-29:
You can export STEP from the KiCAD sources on your own, it is the free software we use to design these boards. Install KiCAD and download the sources from the GitHub page for ESP32-POE (better download the whole repo from this link [ESP32-POE repo](https://github.com/OLIMEX/ESP32-POE/archive/refs/heads/master.zip) ). 

Extract the archive, open the project file in KiCAD, open the PCB layout editor, and click File -> Export and chose one of the formats (STEP in your case). Furthermore, there is a built-in 3D viewer in KiCAD, again open the PCB and navigate to View -> 3D Viewer.

DanKoloff, 2024-10-30:
One thing I forgot to mention is that the component heights in our designs are approximate. Length and width are exact. But always doubt the height of components, since we don't need it for manufacturing. Compare what you get from the STEP with the real board.

closed 2025-01-21 by DanKoloff