OLIMEX/ESP32-POE #55: Why not use 3.3V PG pin as ESP_EN?

marekm72, 2025-04-29:
The simple RC delay may not reset properly for brown-outs or very short power failures (not enough time to fully discharge to 0V, possibly leading to incorrect operation).  But the 3.3V switcher (TPS62A02A) has that nice but currently unused PG pin - why not use it?  It is open drain exactly as needed, just connect it to ESP_EN and it should ensure proper reset in all conditions.

One possible reason why not - if you want the device to work on battery until completely discharged, down to UVLO at 2.3V typ, while PG reset would trigger at about 3.1V.  But it's almost discharged at that point, in this part of discharge characteristics the voltage is already falling quickly so not much run time to gain, and too deep discharge might damage the cell.  If board space permits, the connection from PG to ESP_EN could be made via a small resistor that could be desoldered if necessary.

DanKoloff, 2025-04-30:
Thanks for the suggestion, using the power good pin of TPS62A02A as an extra reset IC is a good idea. We would evaluate and test empirically if that works and will consider it for future revisions.

DanKoloff, 2026-01-20:
A little update - this was successfully tested in latest hardware revision of ESP32-POE-ISO and we would implement it in ESP32-POE in future. Thanks for the idea.