OLIMEX/ESP32-POE #56: Move MPS load to +5VP, optimize for less heat

marekm72, 2025-04-30:
Pulsed MPS with 25% duty cycle would be the best, but if that adds too much complexity - I'd suggest to move the dummy load to the 5V output side.  About 100 ohms parallel to C25 should be fine, looking at the TX4138 efficiency graphs it should sink almost constant 10 mA at the input, and if a bit too low at high voltage then add a little more switching losses by populating C26 and R43 (switching losses increase with input voltage to compensate decreasing current due to constant power).

The current way (in Rev. M - R42 and R7, two 9k1 in parallel at PoE DC/DC input) was made to sink 10 mA at the low end of the voltage range, but power loss rises with voltage squared and the high end of voltage range is the most common case (little voltage drop on the cable at light load), consumes 0.71 W at 57 V when 0.57 W (10 mA) would be enough.

If pulsed MPS is considered - it could actually be done on the secondary side too, and a SOT23-6 MCU like ATtiny4/5 or PIC10F20x could sink the requred current using all 3 of its I/O pins loaded with 3 small resistors, and generate the pulses without any external components.  As an added bonues, the loads could be switched in smaller steps at slightly different times to be nicer to the DC/DC (transient response).  The MCU could also be held in reset (using the remaining input-only pin) to disable the load if not really needed (very likely not needed if the Ethernet PHY is powered up).  This is easy with non-isolated PoE, becomes more complex in the isolated version (but then, the isolated DC/DC converter also contributes some load so the MPS load could be reduced further).

DanKoloff, 2025-04-30:
Thanks for the suggestions. Maybe I miss something but the resistor in parallel to C25 doesn't make a lot of sense since we are moving the heat over the TX4138 and the resistor after it, also there would be much less current available to the board (like 100-150mA less).

We added a second resistor to reduce the heat, it was a single resistor before, we'd probably wait for some customer feedback before doing any changes.

marekm72, 2025-04-30:
Another optimization could be to enable the extra load only when the PHY is powered down - then the current available to the board would not be reduced.  Much (if not all - this would need to be measured) of the MPS required minimum power is already consumed by the Ethernet PHY.  Probably the simplest way could be with N-FET switch, gate controlled from T1 collector, drain via load resistor to +5VP (only powered from PoE, not USB or battery).